eclipse-threadx/samples #3: How to integrate for deliverable?

ccli8, 2020-06-15:
This is continuation of https://github.com/azure-rtos/threadx/issues/8. I've checked the IDE distribute and have the following questions:

1. For a chip vendor to provide azure rtos deliverable, the IDE distribute (including its directory structure) is suggested and necessary?
1. From the IDE distribute, one package for one IDE. How do you manage multi-IDE? It is preferable to keep just one copy of library code (threadx, netduox, etc.) to ease upgrade. Or you have an internal directory structure and do post-process for release?

goldscott, 2020-06-17:
Thanks for your interest in Azure RTOS. We can certainly entertain new packages from chip vendors. We should also discuss licensing opportunities similar to what we are doing with other chip vendors. Please contact azure-rtos-sales@microsoft.com to start a direct communication on contributing new deliverables as well as discussing potential chip vendor licensing.

As for your packaging questions, we suggest separate packages for each supported IDE. As for directory structure, we recommend using something similar to what we have in the current sample packages or leveraging the general product distribution directory structure on Github.

cyliangtw, 2020-06-19:
@goldscott , we are one chip vendor and contacted your local sales one month ago, also understood your business model.
Now, we tried to port on one EVB and consider how to pull request in the future.
So, ask what's the architecture of your main stream of samples, it's for easy merge/rebase in the future. It will be better to realize/follow up your arch.
One more question is where to retrieve the version/commit-hash-code of nx/txd lib in these samples. 
For ex, in [Amazon-FreeRTOS](https://github.com/aws/amazon-freertos), it refer to `freertos_kernel @ 210b1ff` .

goldscott, 2020-06-24:
@cyliangtw This is a good question - i'll find out how the team wants sample projects structured. The sample projects use the 6.0 release of ThreadX and NetX Duo, but because there are IAR sample projects but no IAR ports in the TX and NXD repos, we can't map them to a commit right now. See ports https://github.com/azure-rtos/threadx/tree/master/ports for how each port/toolchain will be structured - we'll have a subdirectory for the architecture and another subdirectory for toolchain.

goldscott, 2020-09-01:
@ccli8 and @cyliangtw - I'll email you privately. Closing.

closed 2020-09-01 by goldscott