eclipse-threadx/samples #4: Question of bin file size (2.87GB) of STM32F746G-DISCO_STM32CubeIDE_Samples

wangyuew, 2020-06-27:
Environment:

- STM32CubeIDE v1.3.1
- Win 7 & 10
- STM32Cube F7 1.16.0

After building these samples ok , i have found there are 3 very large *.bin files.
While the size of bin files of other sample projects is normally little than corresponding elf files.

1. sample_guix_home_automation.bin, 2.87GB.
2. sample_guix_medical.bin, 2.87GB.
3. sample_guix_speedometer.bin, 2.87GB.

I notice the instruction of generating bin file in the console, and i think there is no syntax error.
> arm-none-eabi-objcopy  -O binary  sample_guix_home_automation.elf  "sample_guix_home_automation.bin"

I want solve it by disable the generating of bin file in Project Configuration, but there is no such option in Project --> Properties --> C/C++ Build --> Settings.

yuxin-azrtos, 2020-07-09:
Turns out to be a setting in the linker control file: declare .sd_ram section as (NOLOAD).  We will update the linker file in the future release.  Meanwhile you can replace your local copy (common/STM32F746NGHX_FLASH.ld) using the updated version attached here. Let me know if this fix addresses the issue. 
[STM32F746NGHX_FLASH.zip](https://github.com/azure-rtos/samples/files/4898912/STM32F746NGHX_FLASH.zip)

wangyuew, 2020-07-10:
thanks，its been fixed by the new ld file.

yuxin-azrtos, 2020-07-10:
Thanks!  I am going to close this issue. Feel free to contact us if you have any questions!

closed 2020-07-10 by yuxin-azrtos