eclipse-threadx/samples #10: No rule to make target error

solarflarefx, 2020-10-10:
I am trying to run the sample_azure_iot project and get this same error in stm32cubeide. Any idea what could be wrong? I downloaded the sample directly from: https://github.com/azure-rtos/samples

[image]

Is there some particular step I missed?  I am running on Ubuntu 18.04

wangwen-4220, 2020-10-12:
Hi solarflarefx,

i guess the linker file can't be found. you can check in this way:
[image]

click browse to locate the linker file and try again.

solarflarefx, 2020-10-13:
@wangwen-4220 Thanks for the reply.  How do I access the window you are showing?  Is this a window in STM32CUBEIDE?

solarflarefx, 2020-10-13:
Sorry I found the setting.  It now got past the previous step but I now get these errors:

/opt/st/stm32cubeide_1.4.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.7-2018-q2-update.linux64_1.4.0.202007081208/tools/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lnetxduo
/opt/st/stm32cubeide_1.4.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.7-2018-q2-update.linux64_1.4.0.202007081208/tools/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lthreadx
/opt/st/stm32cubeide_1.4.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.7-2018-q2-update.linux64_1.4.0.202007081208/tools/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lstm32l4xx_lib

wangwen-4220, 2020-10-15:
@solarflarefx ，
It hits that the libraries do not exist, please build netxduo，threadx，lstm32l4xx_lib separately,  then build the sample.

solarflarefx, 2020-10-18:
I tried doing that, but I still get the same error.  Is this an indication the project settings are wrong?

[image]

bo-ms, 2020-10-18:
@solarflarefx we uploaded 6.1 zip projects, could you try the latest if there is same issue?

solarflarefx, 2020-10-18:
@bo-ms I figured it out -- when I downloaded the zip file, it seems that the default paths did not take case into account.  Since Windows paths are not case sensitive, it did not make a difference there but it does in Ubuntu.  The "Debug" directory needed to be capitalized in the paths.  Thanks for the help!

closed 2020-10-18 by solarflarefx

bo-ms, 2020-10-19:
@solarflarefx thanks for bring it up, we will fix this issue.

nik-markovic, 2021-04-20:
@bo-ms This issue is closed, but I was still having the same problem with the latest release Azure_RTOS_6.1_STM32L4+-DISCO_STM32CubeIDE_Samples_2020_10_10.zip. Went to the project, saw no known/open issues and then wasted a lot of time figuring out what is going on. I wish you can either fix this or keep the issue open so folks know about it.

Again, the problems are:

1. Library linker path references in project files are pointing to ../<project_name>/debug, but the build generates upper case  ../<project_name>/Debug directory
2. Linker file is referenced in the project files as ../../common_hardware_code/STM32L4S5VITx_FLASH.ld, but the lunker file name is actually ../../common_hardware_code/STM32L4S5VITX_FLASH.ld (with upper case X).

wangwen-ms, 2021-04-21:
@nik-markovic  thank you for messages , we have fixed this issue internally, and will be released in next version.