eclipse-threadx/samples #11: Azure-Iot-Sample serial port not showing

solarflarefx, 2020-10-18:
I am trying to follow the doc "Azure RTOS IoT Quick Connect Sample STM32L4+-DISCO IoT Node using STM32CubeIDE User Guide".  I was able to get up to the point of downloading the project to the board, but I am having trouble with the following part:
[image]

 I am using Ubuntu 18.04, so I chose to use Putty as the terminal instead.  First I searched for the serial ports using: dmesg | grep tty

I got:
[image]

So within Putty I set up the following:
[image]

However, even after running the program I just get an empty terminal:
[image]

I tried using Azure IoT Explorer's telemetry but it does not seem to communicate:
[image]

wangwen-4220, 2020-10-20:
@solarflarefx ， can you re-plug the USB cable of the board  to make sure the correct serial port is selected?

solarflarefx, 2020-10-21:
@wangwen-4220 I have tested the same serial port on the same board, but using bare metal C code and that works fine.

wangwen-4220, 2020-10-21:
@solarflarefx , want to make sure with you which board you are using , b-l4s5 or b-l475?

solarflarefx, 2020-10-21:
@wangwen-4220  I have the B-L475E-IOT01A1

wangwen-4220, 2020-10-22:
@solarflarefx , please change the linker file to "STM32L475VGTX_FLASH.ld" :

[image]

solarflarefx, 2020-10-29:
Thanks for the suggestions.  Seems to have made a difference, but I get this error now:
[image]

bo-ms, 2020-10-29:
@solarflarefx could you check if set the WIFI user name and password?

solarflarefx, 2020-10-29:
@bo-ms You mean in board_setup.c?

#ifndef WIFI_SSID
//#error "Symbol WIFI_SSID must be defined."
#define WIFI_SSID "xxx"
#endif /* WIFI_SSID  */

#ifndef WIFI_PASSWORD
//#error "Symbol WIFI_PASSWORD must be defined."
#define WIFI_PASSWORD "yyy"
#endif /* WIFI_PASSWORD  */

I edited the lines above, replacing "xxx" and "yyy" with the SSID and password.  Is there anything else I am supposed to edit?

bo-ms, 2020-10-29:
@solarflarefx correct, set WIFI_SSID and WIFI_PASSWORD as yours, could you also check if the WIFI_SECURITY_TYPE is correct? If also failed, could you try to use phone hotspot first?

solarflarefx, 2020-10-29:
@bo-ms I looked at the security settings for my wifi network: 
[image]

My settings were: 
#ifndef WIFI_SECURITY_TYPE
//#error "Symbol WIFI_SECURITY_TYPE must be defined."
#define WIFI_SECURITY_TYPE  WIFI_ECN_WPA2_PSK
#endif /* WIFI_SECURITY_TYPE  */

bo-ms, 2020-10-29:
@solarflarefx seems the settings are correct, one potential issue, the wifi signal on L4 board is weak, could you try to move the router close to board? or try to get the board connected to hotspot? thanks.

solarflarefx, 2020-10-29:
I get this:
STM32L4XX Lib:
> CMSIS Device Version: 1.7.0.0.
> HAL Driver Version: 1.12.0.0.
> BSP Driver Version: 1.0.0.0.
ES-WIFI Firmware:
> Product Name: Inventek eS-WiFi
> Product ID: ISM43362-M3G-L44-SPI
> Firmware Version: C3.5.2.5.STM
> API Version: v3.5.2
ES-WIFI MAC Address: xxx
ES-WIFI Connected.
> ES-WIFI IP Address: x.x.x.x
> ES-WIFI Gateway Address: x.x.x.x
> ES-WIFI DNS1 Address: x.x.x.x
> ES-WIFI DNS2 Address: x.x.x.x
Error in connecting to server: 0x10005

bo-ms, 2020-10-29:
@solarflarefx made some progress, once you get the IP address, it means the device already connected to router. did you set the symbols used to connect IoT Hub in sample_config.h? such as: DEVICE_ID, DEVICE_SYMMETRIC_KEY, HOST_NAME

solarflarefx, 2020-10-29:
@bo-ms I followed the PDF instructions and set:
- HOST_NAME
- DEVICE_ID
- DEVICE_SAS

I did not set other parameters except for the WIFI information.

Is there a way to tell which of the above parameters is giving a problem?

wangwen-4220, 2020-10-30:
@solarflarefx , which sample are you running?
please note that DEVICE_SAS is time-limited. what value did you set for "Expiration (minutes)" when creating SAS token connection string

solarflarefx, 2020-10-30:
@wangwen-4220 I am trying to run sample_azure_iot project.  I set the SAS to 120 minutes.  This time did not expire from when I generated the token and then tried to run the sample.

wangwen-4220, 2020-11-02:
@solarflarefx  ，HOST_NAME's format looks like "your-iothub-name.azure-devices.net". 
how is it going ？ still Error in connecting to server: 0x10005？

solarflarefx, 2020-11-02:
Yes still the same error.  I believe I did put in the correct HOST_NAME

wangwen-4220, 2020-11-02:
can you try the latest release samples : https://github.com/azure-rtos/samples

bo-ms, 2021-01-11:
closing this issue for now, let us know if you have further questions.

closed 2021-01-11 by bo-ms