eclipse-threadx/samples #18: Nothing to build

ShekharSuthar, 2021-02-24:
hello, I'm working on Azure RTOS for the STM32L4 device and I used your repo [Azure_RTOS_6.1_STM32L4+-DISCO_STM32CubeIDE_Samples_2020_10_10.zip](url)

when I read the docs provided in the folder and I tried to build the sample named "sample_azure_iot_embedded_sdk" it gave me this in the console
**Build of configuration Debug for project sample_azure_iot_embedded_sdk
Nothing to build for project sample_azure_iot_embedded_sdk**

I'm using windows 10 system and **STM32CubeIDE 1.4.0** help me out if I'm missing something and ask me if you are looking for some more information

SS-iot-devoted, 2021-02-25:
Hello, Sir 
Is there any update I'm waiting @wangwen-4220

wangwen-ms, 2021-02-26:
@SS-iot-devoted , can you try to clean and build? if still nothing to build, try to turn off virus protection. 
[image]

SS-iot-devoted, 2021-02-26:
hi @wangwen-ms ,
I did what you ask for the clean project and that turn off virus protection. but still getting the same issue
I think the problem is on linker files or path side because in doc _Azure_RTOS_STM32L4+-DISCO_Azure_IoT_Embedded_SDK_For_STM32CubeIDE.pdf_   page10 I got this point   _1. Right click on active sample project, select Properties > C / C++ Build > Settings. In Tool Settings tab, change the linker file to common_hardware_code / STM32L475VGTX_FLASH.ld_

but when I try to set this up I got this error window so can you point out on that side 

[image]

wangwen-ms, 2021-02-26:
can you use the latest stmcube IDE 1.5.1 to open it ?
the linker file has no effect on build.

SS-iot-devoted, 2021-02-26:
I update my IDE ask you asked  and update all other tools we need for that
but still, I'm not able to build it again 

[image]

wangwen-ms, 2021-02-26:
can you try to open the project in this way:
file->import 
[image]

SS-iot-devoted, 2021-02-26:
Still nothing to build...
Getting the same output in console

wangwen-ms, 2021-02-26:
it cant be explained. can you paste your project here , what it looks like when you open it

SS-iot-devoted, 2021-02-26:
I'm working on Azure RTOS for the STM32l475-E-IOT-01-A2 device and I used your repo [https://github.com/azure-rtos/samples/releases/download/v6.1_rel/Azure_RTOS_6.1_STM32L4+-DISCO_STM32CubeIDE_Samples_2020_10_10.zip](url)

when I read the PDF docs provided in the folder (Azure_RTOS_6.1_STM32L4+-DISCO_STM32CubeIDE_Samples_2020_10_10\b-l4s5i-iot01a\stm32cubeide\docs\Azure_RTOS_STM32L4+-DISCO_Azure_IoT_Embedded_SDK_For_STM32CubeIDE.pdf)  there are few steps for setup and run that application .. **Im not able to do that below step  on page10 in pdf doc** 

_1. Right click on active sample project, select Properties > C / C++ Build > Settings. In Tool Settings tab, change the linker file to common_hardware_code / STM32L475VGTX_FLASH.ld_

[image]

I want to make my own application before that I wanna test that sample so I just add Azure configs for connecting the cloud and try to build that the sample named "sample_azure_iot_embedded_sdk" it gave me this in the console

[image]

Build of configuration Debug for project sample_azure_iot_embedded_sdk
Nothing to build for project sample_azure_iot_embedded_sdk

SS-iot-devoted, 2021-03-01:
any update @wangwen-ms ....
I'm still looking for the solution

wangwen-ms, 2021-03-02:
what is your package size downloaded from github, and what does the problem windows show?
[image]

SS-iot-devoted, 2021-03-02:
the zip package is 18.5M. and while building I'm getting this error.

[image]

and in the problem window.

[image]

wangwen-ms, 2021-03-02:
@SS-iot-devoted  please unfold the warnings , what arethe warnings info?

SS-iot-devoted, 2021-03-02:
Description	Resource	Path	Location	Type
make: *** No rule to make target 'clean'.  Stop.	sample_azure_iot_embedded_sdk		 	C/C++ Problem
'%s' directive writing up to 127 bytes into a region of size 40 [-Wformat-overflow=]	platform_STM32Cube_IoT01A1.c	/B-L475E-IOT01/Azure1/User/adapter	line 280	C/C++ Problem
'%s' directive writing up to 127 bytes into a region of size 40 [-Wformat-overflow=]	platform_STM32Cube_IoT01A1.c	/B-L475E-IOT01/Azure1/User/adapter	line 283	C/C++ Problem
'destroyLocalParameter_Bool' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1492	C/C++ Problem
'destroyLocalParameterascii_char_ptr_no_quotes' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1506	C/C++ Problem
'destroyLocalParameterascii_char_ptr' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1497	C/C++ Problem
'destroyLocalParameterdouble' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1457	C/C++ Problem
'destroyLocalParameterEDM_BINARY' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1352	C/C++ Problem
'destroyLocalParameterEDM_BOOLEAN' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1362	C/C++ Problem
'destroyLocalParameterEDM_BYTE' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1367	C/C++ Problem
'destroyLocalParameterEDM_DATE_TIME_OFFSET' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1377	C/C++ Problem
'destroyLocalParameterEDM_DATE' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1372	C/C++ Problem
'destroyLocalParameterEDM_DECIMAL' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1382	C/C++ Problem
'destroyLocalParameterEDM_DOUBLE' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1391	C/C++ Problem
'destroyLocalParameterEDM_DURATION' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1396	C/C++ Problem
'destroyLocalParameterEDM_GUID' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1406	C/C++ Problem
'destroyLocalParameterEDM_INT16' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1411	C/C++ Problem
'destroyLocalParameterEDM_INT32' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1416	C/C++ Problem
'destroyLocalParameterEDM_INT64' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1421	C/C++ Problem
'destroyLocalParameterEDM_SBYTE' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1426	C/C++ Problem
'destroyLocalParameterEDM_SINGLE' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1431	C/C++ Problem
'destroyLocalParameterEDM_STRING' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1436	C/C++ Problem
'destroyLocalParameterEDM_TIME_OF_DAY' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1442	C/C++ Problem
'destroyLocalParameterfloat' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1452	C/C++ Problem
'destroyLocalParameterint' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1447	C/C++ Problem
'destroyLocalParameterint16_t' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1477	C/C++ Problem
'destroyLocalParameterint32_t' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1482	C/C++ Problem
'destroyLocalParameterint64_t' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1487	C/C++ Problem
'destroyLocalParameterint8_t' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1467	C/C++ Problem
'destroyLocalParameterlong' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1462	C/C++ Problem
'destroyLocalParameteruint8_t' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1472	C/C++ Problem
'DeviceTwinCallback' defined but not used [-Wunused-function]	AzureClient_mqtt_DM_TM.c	/B-L475E-IOT01/Azure1/User	line 212	C/C++ Problem
'FromAGENT_DATA_TYPE__Bool' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1124	C/C++ Problem
'FromAGENT_DATA_TYPE_ascii_char_ptr_no_quotes' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1202	C/C++ Problem
'FromAGENT_DATA_TYPE_ascii_char_ptr' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1155	C/C++ Problem
'FromAGENT_DATA_TYPE_double' defined but not used [-Wunused-function]	B-L475E-IOT01		line 834	C/C++ Problem
'FromAGENT_DATA_TYPE_EDM_BINARY' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1315	C/C++ Problem
'FromAGENT_DATA_TYPE_EDM_DATE_TIME_OFFSET' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1250	C/C++ Problem
'FromAGENT_DATA_TYPE_EDM_GUID' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1282	C/C++ Problem
'FromAGENT_DATA_TYPE_float' defined but not used [-Wunused-function]	B-L475E-IOT01		line 866	C/C++ Problem
'FromAGENT_DATA_TYPE_int' defined but not used [-Wunused-function]	B-L475E-IOT01		line 899	C/C++ Problem
'FromAGENT_DATA_TYPE_int16_t' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1029	C/C++ Problem
'FromAGENT_DATA_TYPE_int32_t' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1061	C/C++ Problem
'FromAGENT_DATA_TYPE_int64_t' defined but not used [-Wunused-function]	B-L475E-IOT01		line 1093	C/C++ Problem
'FromAGENT_DATA_TYPE_int8_t' defined but not used [-Wunused-function]	B-L475E-IOT01		line 964	C/C++ Problem
'FromAGENT_DATA_TYPE_long' defined but not used [-Wunused-function]	B-L475E-IOT01		line 931	C/C++ Problem
'FromAGENT_DATA_TYPE_uint8_t' defined but not used [-Wunused-function]	B-L475E-IOT01		line 996	C/C++ Problem
[... cut ...]
in expansion of macro 'MU_FOR_EACH_2_COUNTED_4'	B-L475E-IOT01		line 3322	C/C++ Problem
in expansion of macro 'MU_FOR_EACH_2_COUNTED'	B-L475E-IOT01		line 31	C/C++ Problem
in expansion of macro 'TO_AGENT_DATA_TYPE'	B-L475E-IOT01		line 177	C/C++ Problem
this is the location of the previous definition	B-L475E-IOT01		line 16, external location: c:\st\stm32cubeide_1.5.1\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\include\assert.h	C/C++ Problem

SS-iot-devoted, 2021-03-02:
that warning from another project .. but the problem window  only from this project

goldscott, 2021-03-02:
@wangwen-ms @SS-iot-devoted are you both using the exact same version of Cube IDE and compiler?

Eclipse sucks. Sometimes, even if the vIDE and compiler versions are the same, for whatever reason, projects made on one machine do not work on another machine.

Nobody else has reported any problems opening or building these projects. @SS-iot-devoted can you try on a different machine?

wangwen-ms, 2021-03-03:
@goldscott  yes,  i think so

SS-iot-devoted, 2021-03-03:
Don't know maybe this  is the reason but I tried Azure_RTOS_6.0_STM32L475-DISCO_STM32CubeIDE_Samples_2020_05_17.zip from [https://github.com/azure-rtos/samples/releases](url)
 and here I build that is work for me but getting that error **Error in getting Azure-host address: 0xa3** so can you help me out from here

SS-iot-devoted, 2021-03-03:
Thank you @goldscott for the help that is work for me, I'm using a different system now.

Now I faced another issue which is the application not able to **SNTP Time sync failed** and **IoTHub client connect fail: DNS RESOLVE FAIL status: 163**

[image]

One more thing I want to know that what if I want to use JSON structures which JSON library  I can use and do I need to add from outside or what  ??
Thanks for help

bo-ms, 2021-03-04:
Hi @SS-iot-devoted seems the WIFI firmware (C3.5..2.3.BETA9) is too old, we encountered this issue that the old ES_WIFI did not correctly return the DNS address, could you update the [firmware](https://www.inventeksys.com/iwin/firmware/)?  C6.2.1.7 and C3.5.2.5 has been verified.

bo-ms, 2021-03-04:
In this [branch](https://github.com/azure-rtos/netxduo/tree/feature/iot_pnp/addons/azure_iot), the json (nx_azure_iot_json_reader.c/.h and nx_azure_iot_json_write.c/.h) have been public preview, you can try it.  [Json APIs](https://github.com/azure-rtos/netxduo/blob/feature/iot_pnp/addons/azure_iot/docs/azure_rtos_iot_json.md)

SS-iot-devoted, 2021-03-04:
Hi @bo-ms  For update the [firmware](https://www.inventeksys.com/iwin/firmware/) I got the .bin file for device **B-L475E-IOT01A**. Can you tell me which one I should use(UART/ SPI)  and how can I update the firmware by that .bin file ?

Can I get the .c or .h file so I directly replace in main code

bo-ms, 2021-03-05:
Hi @SS-iot-devoted, could you install the [Inventek ISM 43362 Wi-Fi module firmware update](https://www.st.com/content/ccc/resource/technical/binary_resources/utilities/group0/7d/0b/be/f2/6a/51/4f/a2/inventek_fw_updater/files/inventek_fw_updater.zip/jcr:content/translations/en.inventek_fw_updater.zip)?

SS-iot-devoted, 2021-03-05:
hello @bo-ms ,
I tried what you ask for with the help of the readme file used **STM32 ST-LINK Utility** and its showing that my firmware is updated
[image]

 so I unplug the device and again try to install my application while _RUN_ the application this window popup
[image]

and after again  updating the firmware received **upgread syccesful** 
[image]

my device restart and showing me the same result in the output console window 

[image]

bo-ms, 2021-03-05:
Hi @SS-iot-devoted from the screenshot, the firmware version is still C3.5.2.3.BETA9. Did you run .bat?

SS-iot-devoted, 2021-03-05:
I didn't post every screenshot but if you want I can post it here all with all steps.
Yes, I run **update_Wifi_FW_C3.5.2.5.STM.bat** file at the end of the 7th step
here the results that I got ...   Can we make a meeting on **Skype or MSTeam** ???
.......................................................................

C:\WINDOWS\system32>"C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility\ST-LINK_CLI.exe" -c -me -p .\InventekBootloaderPassthrough.bin 0x08000000 -v -rst -hardrst
STM32 ST-LINK CLI v3.5.0.0
STM32 ST-LINK Command Line Interface

ST-LINK SN: 066EFF485450827867183610
ST-LINK Firmware version: V2J34M25
Connected via SWD.
SWD Frequency = 4000K.
Target voltage = 3.2 V
Connection mode: Normal
Reset mode: Hardware reset
Device ID: 0x415
Device flash Size: 1024 Kbytes
Device family: STM32L4x1/L4x5/L4x6

**Full chip erase...
Flash memory erased.**

**Loading file...
Unable to open file!**

MCU Reset.

Hard reset is performed.

C:\WINDOWS\system32>timeout /t 10

Waiting for  3 seconds, press a key to continue ...

C:\WINDOWS\system32>"C:\Program Files (x86)\STMicroelectronics\Software\Flash Loader Demo\STMFlashLoader.exe" -c --pn 5 --br 115200  --db 8 --pr EVEN --sb 1 --to 30000 -i STM32F2_1024K -e --sec 10 0 1 2 3 4 5 6 7 8 9 -d --fn .\ISM43362_M3G_L44_SPI_C3.5.2.5.STM.bin --a 0x08000000 --v
Opening Port                             [OK]
Activating device                                [KO]
No response from the target, the Boot loader can not be started.
Please, verify the boot mode configuration, reset your device then try again.
Please, reset your device then press any key to continue

 Press any key to continue ...

Opening Port                             [OK]
Activating device                                [KO]
No response from the target, the Boot loader can not be started.
Please, verify the boot mode configuration, reset your device then try again.
Please, reset your device then press any key to continue

 Press any key to continue ...

bo-ms, 2021-03-05:
you can check the current wifi firmware version in the output console window. 
[image]

SS-iot-devoted, 2021-03-05:
It is still the same as that image

bo-ms, 2021-03-05:
Looks the target is not connected. let's chat. My MS Teams: boche@microsoft.com

SS-iot-devoted, 2021-03-05:
Thank @bo-ms for helping me now seems working for me 
thanks again

SS-iot-devoted, 2021-03-08:
hello @bo-ms 

I just wanna know that how can I use the other branch libraries in that **sample_azure_iot_embedded_sdk** project 
I want to use JSON and RTC so

bo-ms, 2021-03-08:
The JSON is already included in the sample project, see [netxduo\addons\azure_iot\nx_azure_iot_json_reader/writer.*](https://github.com/azure-rtos/netxduo/tree/master/addons/azure_iot). Here is [JSON APIs](https://github.com/azure-rtos/netxduo/blob/master/addons/azure_iot/docs/azure_rtos_iot_json.md)

SS-iot-devoted, 2021-03-09:
hello @bo-ms these all application are based on treading even if I want to use need to use the treading concept  
can I do this Azure RTOS as with a normal C console application based?

actually, I need to build my own application that has several steps to connect Azure and so on another task soo

bo-ms, 2021-03-11:
@SS-iot-devoted Yes, you can change the steps of connecting Azure in you own application. Closing this issue. We can make a meeting if needed.

closed 2021-03-11 by bo-ms