eclipse-threadx/samples #22: STM32L4+ CubeIDE: Legacy project names causing linker issues and include path confusion

nik-markovic, 2021-04-27:
This issue existis with Azure_RTOS_6.1_STM32L4+-DISCO_STM32CubeIDE_Samples_2020_10_10.zip

I suspect that netxduo/threadx/stm32l4xx_libwere once named nxd/tx/stm32l475_lib respectively. This left some residue in .cproject files that cause issues if a new project gets created with STM32CubeIDE and references netxduo/threadx/stm32l4xx_lib.

Besides extra and incorrect non-existent include paths paths in the newly created project, the linker will fail unable to find nxd/tx/stm32l475_lib because linker flags would be included adding -ltx -lnxd and -lstm32l475_lib.

The following sed substitutions partially resolve these issues, but I would suggest to remove .project and .cproject files and recreating those projects from scratch:

```shell script
sed -i '/stm32l475_lib/d' ./stm32l4xx_lib/.cproject
sed -i '/nxd/d' ./netxduo/.cproject
sed -i '/stm32l475_lib/d' ./threadx/.cproject
sed -i 's#\/tx#\/threadx#g' ./threadx/.cproject
sed -i 's#"tx"#"threadx"#g' ./threadx/.cproject
```

wangwen-ms, 2021-04-29:
thanks for the suggestion, 
we will create new project in the l4 project sand evaluate if the changed names cause such issue.

nik-markovic, 2021-04-29:
@wangwen-ms The changed names also seem to be affecting the NXP 1060 MCU Expresso project. In C/C++ build -> Refresh policy the item name is "tx" instead of "threadx", which causes a full rebuild instead of incremental.

wangwen-ms, 2021-05-07:
thanks for bring this up, we will evaluate the consequence caused by changed names and see how to fix it.

wangwen-ms, 2021-06-25:
we will fixed it in next release.

bo-ms, 2021-07-12:
Closing it since we already fixed it internally and this bug fix will be in next release.

closed 2021-07-12 by bo-ms