eclipse-threadx/samples #27: MIMXRT1060 FlexRAM adjustment and debug

faceface1234, 2021-09-03:
Hi,
1. We are working on a project based on the sample_azure_iot_embedded_sdk_adu project of MIMXRT1060. Our board is MIMXRT1060-EVK.
https://github.com/azure-rtos/samples/releases/download/rel_6.1_adu_beta/Azure_RTOS_6.1_ADU_MIMXRT1060_MCUXpresso_Sample_2021_03_02.zip
2. We followed the instructions below and tried to increase the SDRAM_DTC from 128KB to 256KB by decreasing the OC_SDRAM. The GPR17 register value is set to 0x5AAFFAA5. 
https://community.nxp.com/t5/i-MX-RT-Knowledge-Base/Reallocating-the-FlexRAM/ta-p/1117649

3. The issue is that when debugging, heap and stack usage seem normal, but it fails at HardFault_Handler when triggering memory related functions, like malloc and srand. 
Attached is the connection script that we use for debugging and the default hard-coded linker script used by the project(moved the stack from end to start). 

Thanks,
Face
[connect_debug.txt](https://github.com/azure-rtos/samples/files/7103536/connect_debug.txt)
[MIMXRT1062xxxxx_flexspi_nor_ota.txt](https://github.com/azure-rtos/samples/files/7103537/MIMXRT1062xxxxx_flexspi_nor_ota.txt)

Tim-Wang38, 2021-09-03:
Hi,
Do you modify the MPU configuration to adapt the modification of flexram?
[image]

faceface1234, 2021-09-03:
Yes, I did. For RT1060, there are 512KB reserved for SRAM_OC. 

    /* Region 4 setting: Memory with Normal type, not shareable, outer/inner write back */
    MPU->RBAR = ARM_MPU_RBAR(4, 0x00000000U);
    MPU->RASR = ARM_MPU_RASR(0, ARM_MPU_AP_FULL, 0, 0, 1, 1, 0, ARM_MPU_REGION_SIZE_128KB);

    /* Region 5 setting: Memory with Normal type, not shareable, outer/inner write back */
    MPU->RBAR = ARM_MPU_RBAR(5, 0x20000000U);
    MPU->RASR = ARM_MPU_RASR(0, ARM_MPU_AP_FULL, 0, 0, 1, 1, 0, ARM_MPU_REGION_SIZE_256KB);

    /* Region 6 setting: Memory with Normal type, not shareable, outer/inner write back */
    MPU->RBAR = ARM_MPU_RBAR(6, 0x20200000U);
    MPU->RASR = ARM_MPU_RASR(0, ARM_MPU_AP_FULL, 0, 0, 1, 1, 0, ARM_MPU_REGION_SIZE_512KB);

    /* Region 7 setting: Memory with Normal type, not shareable, outer/inner write back */
    MPU->RBAR = ARM_MPU_RBAR(7, 0x20280000U);
    MPU->RASR = ARM_MPU_RASR(0, ARM_MPU_AP_FULL, 0, 0, 1, 1, 0, ARM_MPU_REGION_SIZE_128KB);

goldscott, 2021-09-03:
Please contact NXP for help with this issue.

Tim-Wang38, 2021-09-05:
Hi,

I've noticed that you set the heap to the SDRAM, did you enable the macro 'XIP_BOOT_HEADER_DCD_ENABLE' in the predefine configuration?

faceface1234, 2021-09-09:
> Hi,
> 
> I've noticed that you set the heap to the SDRAM, did you enable the macro 'XIP_BOOT_HEADER_DCD_ENABLE' in the predefine configuration?

Yes I did try to set it to 1. Program still failed at the HardFault_Handler().

faceface1234, 2021-09-10:
I found that I might need to change the RAM/Boot sections in the ADU project's hardcoded linker file. I tried to compared the auto-generated and hardcoded linker files(attached), but I could not understand much. Can you please take a look at them and see if there are any problems? 

[MIMXRT1062xxxxx_flexspi_nor_ota.txt](https://github.com/azure-rtos/samples/files/7140554/MIMXRT1062xxxxx_flexspi_nor_ota.txt)
[sample_azure_iot_embedded_sdk_adu_OTA_Debug.txt](https://github.com/azure-rtos/samples/files/7140555/sample_azure_iot_embedded_sdk_adu_OTA_Debug.txt)
[sample_azure_iot_embedded_sdk_adu_OTA_Debug_library.txt](https://github.com/azure-rtos/samples/files/7140556/sample_azure_iot_embedded_sdk_adu_OTA_Debug_library.txt)
[sample_azure_iot_embedded_sdk_adu_OTA_Debug_memory.txt](https://github.com/azure-rtos/samples/files/7140557/sample_azure_iot_embedded_sdk_adu_OTA_Debug_memory.txt)

wangwen-ms, 2021-09-10:
> Please contact NXP for help with this issue.

Hi Scott,

 i forwarded this issue to @Tim-Wang38  who comes from NXP.

goldscott, 2022-09-06:
Thank you @Tim-Wang38 for helping! I didn't realize you were from NXP. @faceface1234 - have you resolved this issue?

faceface1234, 2022-09-12:
Scott,

We use another board instead.

Face

closed 2022-11-16 by liydu