eclipse-threadx/samples #29: GUIX

rploeg, 2021-10-31:
Hello,

I am using the sample for Azure RTOS - STM32F476G-DISCO

Azure_RTOS_6.1_STM32F746G-DISCO_STM32CubeIDE_Samples_2020_10_10.zip

Everything I can upload to my device and it's working. But I have a question regarding editing the screens. Is it possible to chance the screens of the app with GUIX Studio?

ting-ms, 2021-11-01:
Yes you can edit the GUIX Studio project(.gxp file) that contains in the sample.

rploeg, 2021-11-19:
Thank you. I have changed the file, but after a rebuild of the project in the IDE workspace of STM and deploy to the device, still the old screens are shown. When I open the GUIX Studio I see the new screens. Do I need to rebuild on another way to include the new screens?

rploeg, 2021-11-19:
Sorry found it, should generate output files in the GUIX Studio.

closed 2021-11-19 by rploeg