eclipse-threadx/samples #30: Win32 sample - Access Violation Error

tolgayilmaz86, 2021-11-01:
In the below line I get "Unhandled exception at 0x74EBB1AF (combase.dll) in ProjectHelloWorld.exe: 0xC0000005: Access violation reading location 0x00000004. occured"

tx_thread_schedule.c line 156
`tx_thread_schedule(VOID)::WaitForSingleObject(_tx_win32_scheduler_semaphore, INFINITE);`

The project is cloned and built successfully and no change has been applied.

liydu, 2021-11-09:
Hi @tolgayilmaz86 , can we know which version of Visual Studio were you using? I just tried on the latest VS2022 (used to test on VS2019) it launched okay. Thanks!

tolgayilmaz86, 2021-11-09:
Sure, I was on the VS2019 (&Windows 10) but it seems it worked on its own without changing anything. Maybe some conflict with my system but could be something to consider within code side just to prevent the issue.

liydu, 2021-11-10:
@tolgayilmaz86 thanks for the further info. We will track this as an improvement.

closed 2021-11-10 by tolgayilmaz86