eclipse-threadx/samples #33: Missing Documentation Steps for .bin file and Powershell Script in SAME54 ADU Sample

t-tbrown, 2022-08-04:
Within the `same54Xpro\mplab\docs\Azure_RTOS_ATSAME54-XPRO_Device_Update_For_MPLAB.pdf` file for the documentation, in the **Build new **firmware****, there is a section to generate the `.bin` firmware binary and use that binary in the update manifest.

[image]

In the default MPLAB X IDE, the binaries are only generated as HEX, ELF, and MAP files. In order to generate the binary as .bin, you have to change some settings in the _post build commands_ settings in MPLAB IDE, like shown here: [MPLAB Forum](https://microchipsupport.force.com/s/article/MPLAB-X-IDE---Generating-binary-file---xc32-Compiler#:~:text=Answer.%20MPLAB%20X%20does%20not%20create%20.bin%20file,build%20%E2%80%99%20option%20and%20insert%20the%20following%20command.)

Could this be included in the documentation so users don't have to find this forum post themselves to fix the issue?

Additionally, I am running Windows 1110.0.22000. When I try to follow the instruction in **Generate Import Manifest Section** of the documentation, I run into an error in PowerShell v7.0 (admin) when running `Set-ExecutionPolicy -ExecutionPolicy RemoteSigned -Scope Process`

```
> Set-ExecutionPolicy -ExecutionPolicy RemoteSigned -Scope Process
> .\CreateSAME54Update.ps1
.\CreateSAME54Update.ps1: File CreateSAME54Update.ps1 cannot be loaded. The file CreateSAME54Update.ps1 is not digitally signed. You cannot run this script on the current system. For more information about running scripts and setting execution policy, see about_Execution_Policies at https://go.microsoft.com/fwlink/?LinkID=135170.
```

I worked around this error by using `Set-ExecutionPolicy -ExecutionPolicy Bypass -Scope Process`.

goldscott, 2022-08-04:
I think this is a copy-paste error in the documentation because that path is for the MCUXpresso tools, not the MPLAB tools. @wangwen-4220 can you look into this?

t-tbrown, 2022-08-04:
> I think this is a copy-paste error in the documentation because that path is for the MCUXpresso tools, not the MPLAB tools. @wangwen-4220 can you look into this?

Sorry @goldscott , had both documents open and copied the wrong one. I updated the issue with the correct screenshot.

wangwen-ms, 2022-08-05:
Hi @t-tbrown, we will fix the binary file generating in next release. thanks.

bo-ms, 2022-08-05:
Hi @t-tbrown thanks for reporting the policy issue, and great you have a workaround for it, we will update the document for it. Much thanks.

liydu, 2022-12-01:
@t-tbrown thanks for pointing out and it has been updated in the latest v6.2 sample projects: https://github.com/azure-rtos/samples/releases/tag/v6.2_rel

closed 2022-12-01 by liydu