eclipse-threadx/samples #36: Build fail for STM32 B-L75E sample_azure_iot_embedded_sdk

kae-made, 2023-06-27:
I'm tring https://github.com/azure-rtos/samples/releases/download/v6.2_rel/Azure_RTOS_6.2_STM32L4+-DISCO_STM32CubeIDE_Samples_2022_11_30.zip.
I've worked according to Azure_RTOS_STM32L4+-DISCO_Azure_IoT_Embedded_SDK_For_STM32CubeIDE.pdf.
This document describe that

> The sample project can also run on STM32L475-DISCO IoT Node with just a 
few configurations:
1. Right click on active sample project, select Properties > C / C++ Build 
> Settings. In Tool Settings tab, change the linker file to
common_hardware_code / STM32L475VGTX_FLASH.ld

The board I'm using STM32 B-L475E so I do this selection and build.
But build error occurs.
```
09:45:02 **** Incremental Build of configuration Debug for project sample_azure_iot_embedded_sdk ****
make -j8 all 
arm-none-eabi-gcc -o "sample_azure_iot_embedded_sdk.elf" @"objects.list"  -lnetxduo -lthreadx -lstm32l4xx_lib -mcpu=cortex-m4 -T"C:\demo\Azure_RTOS_6.2_STM32L4+-DISCO_STM32CubeIDE_Samples_2022_11_30\b-l4s5i-iot01a\stm32cubeide\common_hardware_code\STM32L475VGTX_FLASH.ld" --specs=nosys.specs -Wl,-Map="sample_azure_iot_embedded_sdk.map" -Wl,--gc-sections -static -L../../threadx/Debug -L../../netxduo/Debug -L../../stm32l4xx_lib/Debug --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -Wl,--start-group -lc -lm -Wl,--end-group
c:\st\stm32cubeide_1.12.1\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.200.202301161003\tools\arm-none-eabi\bin\ld.exe: sample_azure_iot_embedded_sdk.elf section `.bss' will not fit in region `RAM'
c:\st\stm32cubeide_1.12.1\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.200.202301161003\tools\arm-none-eabi\bin\ld.exe: region `RAM' overflowed by 22632 bytes
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:63: sample_azure_iot_embedded_sdk.elf] Error 1
"make -j8 all" terminated with exit code 2. Build might be incomplete.

09:45:05 Build Failed. 3 errors, 0 warnings. (took 3s.421ms)
```

When I select STM32L4SSVITX_FLASH.ld, this build success.
Please let me know how to fix this problem.

wangwen-4220, 2023-06-27:
@kae-made, The application size is over the stm32l475 RAM size, could you add "#define SAMPLE_PACKET_COUNT        (16)" in nx_port.h to reduce the application size to fit stm32l475?  then it can be compiled with stm32l475 linker file.
We will remove the document part of STM32L475-DISCO in next sample project release, thanks for bring it up.

kae-made, 2023-06-28:
@wangwen-4220 , thank you for good information.
I added SAMPLE_PACKET_COUNT in nx_port.h and build became success.
Next step, I try to run this code. 
Monitoring log is as follows.
```
STM32L4XX Lib:
> CMSIS Device Version: 1.7.0.0.
> HAL Driver Version: 1.12.0.0.
> BSP Driver Version: 1.0.0.0.
ES-WIFI Firmware:
> Product Name: Inventek eS-WiFi
> Product ID: ISM43362-M3G-L44-SPI
> Firmware Version: C3.5.2.3.BETA9
> API Version: v3.5.2
ES-WIFI MAC Address: C4:7F:51:5:41:5A
wifi connect try 1 times
ES-WIFI Connected.
> ES-WIFI IP Address: 192.168.68.70
> ES-WIFI Gateway Address: 192.168.68.1
> ES-WIFI DNS1 Address: 255.255.255.255
> ES-WIFI DNS2 Address: 255.255.255.255
IP address: 192.168.68.70
Mask: 255.255.252.0
Gateway: 192.168.68.1
DNS Server address: 255.255.255.255
SNTP Time Sync...0.pool.ntp.org
SNTP Time Sync...1.pool.ntp.org
SNTP Time Sync...2.pool.ntp.org
SNTP Time Sync...3.pool.ntp.org
SNTP Time Sync...0.pool.ntp.org
SNTP Time Sync...1.pool.ntp.org
SNTP Time Sync...2.pool.ntp.org
SNTP Time Sync...3.pool.ntp.org
```
It seems that DNS server initialization failed.
Is this same problem of https://github.com/azure-rtos/getting-started/issues/388 ?

wangwen-4220, 2023-06-28:
Yes, please give it a try.

kae-made, 2023-07-01:
I confirmed that [B-L4S5I-IOT01A](https://www.st.com/en/evaluation-tools/b-l4s5i-iot01a.html) app run on STM32 B-L475.
The cause of DNS client problem was the same as https://github.com/azure-rtos/getting-started/issues/388.
It's OK to close this issue.

Thank you!